pothosware/SoapyPlutoSDR #21: Support for DC Block Option

unixpunk, 2019-03-16:
Hi, thanks for the work here and with SoapySDR in general!  I'm looking for some guidance on options I can (or can't) pass through using rx_sdr's new -t option.  I'm looking to do DC spike removal and offset tuning.  Thanks in advance!

unixpunk, 2019-03-16:
[-t SDR settings (ex: rfnotch_ctrl=false,dabnotch_ctrlb=true)]

unixpunk, 2019-03-18:
I tried pass some of these options and did not see a change in the result in the output running again while an rx stream was active:

# iio_attr -q -i -c ad9361-phy voltage2
dev 'ad9361-phy', channel 'voltage2' (input), attr 'bb_dc_offset_tracking_en', value '1'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'filter_fir_en', value '1'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'gain_control_mode_available', value 'manual fast_attack slow_attack hybrid'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'offset', value '57'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'quadrature_tracking_en', value '1'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'raw', value '1072'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'rf_bandwidth', value '200000'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'rf_bandwidth_available', value '[200000 1 56000000]'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'rf_dc_offset_tracking_en', value '1'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'rf_port_select_available', value 'A_BALANCED B_BALANCED C_BALANCED A_N A_P B_N B_P C_N C_P TX_MONITOR1 TX_MONITOR2 TX_MONITOR1_2'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'sampling_frequency', value '1249999'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'sampling_frequency_available', value '[520833 1 61440000]'
dev 'ad9361-phy', channel 'voltage2' (input), attr 'scale', value '0.305250'

zuckschwerdt, 2019-03-18:
You are spot on. The API isn't supported in SoapySDR. I'll add the support now.

zuckschwerdt, 2019-03-18:
Would you be looking for attributes in other chips/channels besides `ad9361-phy` `voltage2`? Not sure which values to pass through as device default

zuckschwerdt, 2019-03-18:
s.a. #22

guruofquality, 2019-03-18:
Typically there is a hasDCOffsetMode() an set/getDCOffsetMode(). If there is a rx dc removal, then set/get is missing: https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Settings.cpp#L192

zuckschwerdt, 2019-03-18:
The PlutoSDR has

	iio:device1: ad9361-phy
			voltage0:  (input)
				attr  0: bb_dc_offset_tracking_en value: 1
				attr  9: rf_dc_offset_tracking_en value: 1
			voltage2:  (input)
				attr  0: bb_dc_offset_tracking_en value: 1
				attr  8: rf_dc_offset_tracking_en value: 1

I'm not sure what those attributes control though?

zuckschwerdt, 2019-03-18:
Work in #22 is very early but should already give access to:

* Device
  * Calibration Mode [key=calib_mode, default=auto, type=string, options=(auto, manual, manual_tx_quad, tx_quad, rf_dc_offs, rssi_gain_step)]
  * Ensm Mode [key=ensm_mode, default=fdd, type=string, options=(sleep, wait, alert, fdd, pinctrl, pinctrl_fdd_indep)]
  * TRX Rate Governor [key=trx_rate_governor, default=nominal, type=string, options=(nominal, highest_osr)]
  * XO Correction [key=xo_correction, default=40000000, type=string, range=[3.9992e+07, 4.0008e+07, 1]]

* RX Channel
  * bb_dc_offset_tracking_en [key=bb_dc_offset_tracking_en, type=bool]
  * filter_fir_en [key=filter_fir_en, type=bool]
  * quadrature_tracking_en [key=quadrature_tracking_en, type=bool]
  * rf_dc_offset_tracking_en [key=rf_dc_offset_tracking_en, type=bool]

* TX Channel
  * filter_fir_en [key=filter_fir_en, type=bool]

unixpunk, 2019-03-18:
Thanks!  After some more testing this weekend, I'm not even sure if/how these options will help in my ultimate need to avoid the dc spike because they're all always 1 anyway when using rx_sdr...I'm not sure if its the offset value that needs changing or if there is another device aside from the 9361 that might have other options.  `SoapySDRUtil --probe` shows the pluto as having DC Removal, but not really sure what that means, i.e., is it hw-based?  (I can find no evidence of that, and by the way my question here was answered, sounds like it has to be done in sw or an offset needs to be changed along with the dc offset options being set to 1?  https://ez.analog.com/university-program/f/q-a/108619/where-to-find-plutosdr-specific-driver-level-settings-i-e-dc-block )

zuckschwerdt, 2019-03-18:
Try `iio_info -a` and look around the attributes if something interesting stands out. The attributes listed above are the ones I found to perhaps be useful. Both mentioned in your linked question are available, but anyway set to enabled by default (at least for me).

The reported DC Removal is a fixed information not a reading.

unixpunk, 2019-04-05:
Sorry for delay in testing..I'm not actually seeing the option stick on the device.

For example: `rx_sdr -f 98700000 -g 73 -t filter_fir_en=0 /dev/null` still shows 'filter_fir_en', value '1' in iio_attr.

Is there a way I can test this without rx_sdr to confirm this portion is good? Then if need-be I can open an issue for rx_tools.

zuckschwerdt, 2019-12-03:
Please reopen if there is again interest in this and more testing has been done.

closed 2019-12-03 by zuckschwerdt