pothosware/SoapyPlutoSDR #31: Sample rates between

csete, 2019-12-29:
Hi,

I noticed that sample rates between 25e6/48 and 25e6/12 don't work. I found the answer in the [Linux IIO driver notes](https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361), which states that:

> The minimum ADC rate is 25MSPS. Baseband rates below 2.083 MSPS (25MSPS/12) require FIR decimation/interpolation to be set. In other words the FIR filter needs to be configured and enabled. The minimum baseband rate with the FIR filter (decimate by 4) enabled is: 25MSPS /(4*12) = 520.83 kSPS.

I think it requires manually setting FIR filter taps, though I am not sure. I didn't have time to play around with it.

Rates below 25e6/48 seem to work the way it is currently handled in the code.

zuckschwerdt, 2020-02-21:
If I read that right we have three decimation/interpolation stages and the FIR stage.
The decimation/interpolation can be 2,bypass at two stages and one stage with 3,2,bypass.
The FIR can be 4,2,1,bypass. The ADC/DAC minimum rate is 25 MHz. 
Without enabling the decimation/interpolation of the FIRs we can go down to 25M/12 = 2.083 MSps. But we currently only scale by 8, thus should only be able to go down to 25M/8 = 3.125Msps?
Likely, we should scale by 12 if below 25M/8 and load some FIRs if below 25M/12.

closed 2020-03-03 by zuckschwerdt

mndza, 2020-03-25:
Hi,

I think there is a some missing information here: when `libad9361-iio` is available the FIR filter taps are automatically loaded when setting sample rates down to 25e6/48 by calling `ad9361_set_bb_rate`. Otherwise, these sample rates need the user to set the taps manually.

zuckschwerdt, 2020-03-25:
@csete were you missing libad9361-iio in your tests?

csete, 2020-03-28:
@zuckschwerdt I probably did. Should libad9361-iio be installed before libiio or just before SoapyPlutoSDR?

csete, 2020-03-28:
I meant to say "I probably did not have libad9361-iio installed"

zuckschwerdt, 2020-03-28:
It just needs to be around when SoapyPlutoSDR is compiled. `cmake` should show e.g. 
```
Found libad9361: /usr/local/lib/libad9361.dylib (found version "0.2")
```
Could you test e5313e0 again (`git checkout e5313e0`)? Current master circumvents the FIR from libad9361 :/

csete, 2020-03-29:
Thanks for the clarification. I'm still a bit confused about what is lower sample rate limit without libad9361-iio, though it seems that I can now go below 25e6/48 without having libad9361-iio installed.

Here is the output of SoapyPlutoSDR cmake (latest master):

> alc@solaris:~/sdr/src/soapy-plutosdr.git/build$ cmake .. -L
> -- LIBIIO_INCLUDE_DIRS: /home/alc/sdr/runtime/include
> -- LIBIIO_LIBRARIES: /home/alc/sdr/runtime/lib/libiio.so
> -- LIBIIO_DEFINITIONS: 
> -- Could NOT find libad9361 (missing: LIBAD9361_LIBRARY LIBAD9361_INCLUDE_DIR) 
> -- Module PlutoSDRSupport configured with version: 0.2.0-c88b7f5
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/alc/sdr/src/soapy-plutosdr.git/build
> -- Cache values
> CMAKE_BUILD_TYPE:STRING=release
> CMAKE_INSTALL_PREFIX:PATH=/home/alc/sdr/runtime
> LIB_SUFFIX:STRING=
> SOAPY_SDR_INCLUDE_DIR:PATH=/home/alc/sdr/runtime/include
> SOAPY_SDR_LIBRARY:FILEPATH=/home/alc/sdr/runtime/lib/libSoapySDR.so
> SoapySDR_DIR:PATH=/home/alc/sdr/runtime/share/cmake/SoapySDR

Both the SoapySDRUtil transfer test and the spectrum shown with a simple GNU Radio 3.8 / gr-soapy test script shows that I can use both 1, 2, and 3 MSPS with the current master

1 MSPS:
[image]

2 MSPS:
[image]

3 MSPS:
[image]

csete, 2020-03-29:
And, by the way, thanks for the fix!

zuckschwerdt, 2020-03-29:
Ok, the fix works but isn't great if we have libad9361. I'll make it conditional to use the old behaviour when libad9361 is there, otherwise the current one.

commit c88b7f5b (2020-03-03): Fix sample rates between 25e6/48 and 25e6/12 (closes #31)