pothosware/SoapyPlutoSDR #34: cmake not finding the soapysdr directory

waterwin, 2020-08-09:
Building in a Ubuntu 18.04 VirtualBox on Solus Linux. Solus host, Ubuntu client.
Built all Sophos and Soapy materials and libiio and libad9361

But cmake throws error that SoapySDR can not be found.
Did not change any paths from the standard install.

I will now try SoapySDRplay install to see if same is happening.

guruofquality, 2020-08-09:
Install soapysdr first

* https://github.com/pothosware/SoapySDR/wiki/BuildGuide#get-the-source-code
* https://github.com/pothosware/SoapySDR/wiki/BuildGuide#build-and-install

waterwin, 2020-08-10:
It even looks as if I did not install SoapySDR at all (my bad).
I have now installed SoapySDR and running cmake .. from ~/SoapyPlutoSDR/build has new errors.
libiio not found. I did install ;ibiio and libad9361 from the deb packages on the AD website for Ubuntu Bionic Beaver(v 18.04) which was handled via the standard Ubuntu package manager. As that is where the 2 links for these 2 dependencies take me.

libad9361.so is in /usr/local/lib whereas libiio is in /usr/lib/x86_64-linux-gnu

I will now build these 2 from source to see if that makes any difference.
Or maybe should try an install with gdebi first.
Or try one of the other simpler modules first like the RTL-SDR
As currently: SoapySDRUtil --info gives
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
No modules found!
There even is no SoapySDR folder there.
--

waterwin, 2020-08-10:
After build from source of SoapySDR and Pothos and the modules all without errors or warnings. Not found in Pothos GUI, both devices not found usable in GQRX, no blocks for these 2 devices in GNUradio. I am at a loss now. Pothos test is 131 x success. All other tests see below, all looking good to me.
One path is modules0.8 (install) the other modules0.7 (search) which is odd (not Ood).

SoapySDRUtil --info

##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-ga8df1c57
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libPlutoSDRSupport.so (0.2.0-472e9a0)
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so  (0.3.0-8c4e330)
Available factories... plutosdr, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

PothosUtil --system-info
Lib Version: 0.7.0-ge2f336a4
API Version: 0.7.0
ABI Version: 0.7
Root Path: /usr/local
Data Path: /usr/local/share/Pothos
User Data: /home/erwin/.local/share/Pothos
User Config: /home/erwin/.config/Pothos
Runtime Library: /usr/local/lib/libPothos.so.0.7.0
Util Executable: /usr/local/bin/PothosUtil
Dev Include Path: /usr/local/include
Dev Library Path: /usr/local/lib
Module Search Paths:
 * /usr/local/lib/Pothos/modules0.7

Both connect via usb
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1df7:3000  (this is the SDRplay, this is normal)
Bus 001 Device 004: ID 0456:b673 Analog Devices, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

SoapySDRUtil --probe="driver=plutosdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=plutosdr

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9364
  ad9361-phy,xo_correction=39999857
  backend_version=0.21 (git tag: v0.21  )
  fw_version=v0.32
  hw_model=Analog Devices PlutoSDR Rev.B (Z7010-AD9364)
  hw_model_variant=0
  hw_serial=1044739659930004feff1f009e1128ef96
  library_version=0.21 (git tag: ef12218)
  local,kernel=4.19.0-119999-g6edc6cd
  uri=usb:1.3.5
  usb,idProduct=b673
  usb,idVendor=0456
  usb,libusb=1.0.21.11156
  usb,product=PlutoSDR (ADALM-PLUTO)
  usb,release=2.0
  usb,serial=1044739659930004feff1f009e1128ef96
  usb,vendor=Analog Devices Inc.
(etc)

guruofquality, 2020-09-01:
Cool!

closed 2020-09-01 by guruofquality