pothosware/SoapyPlutoSDR #35: Segmentation fault on any radio program

Ev1lbl0w, 2020-09-13:
I'm using an ADALM-PLUTO device, and have installed this driver to use with programs such as **SigDigger**. However, I get a segmentation fault in those programs. I have also tried the probe command `SoapySDRUtil --probe="driver=plutosdr"`, but no luck.

Looking at the core dump and by compilling some libraries, I found out that the code responsible for finding devices *(in [PlutoSDR_Settings.cpp](https://github.com/pothosware/SoapyPlutoSDR/blob/472e9a09b80e15148383efda3702773548c7962c/PlutoSDR_Settings.cpp#L35))* is returning null values:

```
dev = iio_context_find_device(ctx, "ad9361-phy");
rx_dev = iio_context_find_device(ctx, "cf-ad9361-lpc");
tx_dev = iio_context_find_device(ctx, "cf-ad9361-dds-core-lpc");
this->setAntenna(SOAPY_SDR_RX, 0, "A_BALANCED");
this->setAntenna(SOAPY_SDR_TX, 0, "A");
```

Both `dev`, `rx_dev` and `tx_dev` are initialized to 0x00, causing the crash. I initially assumed it was a problem with the **ad9361** driver, but after opening an issue there (https://github.com/analogdevicesinc/libad9361-iio/issues/56) it was suggested I ask here instead.

I'm using Arch Linux, and installed [soapyplutosdr](https://www.archlinux.org/packages/community/x86_64/soapyplutosdr/) which should include everything needed AFAIK.

Probably a duplicate of #25, but when I compiled this I had zero warnings.

zuckschwerdt, 2020-09-13:
If all `dev` are `NULL` that suggests that `ctx` wasn't usable. But it obviously wasn't null, there is a check for that.
So you have a valid context (`ctx`) that doesn't have "ad9361-phy", "cf-ad9361-lpc",  "cf-ad9361-dds-core-lpc"?
Can you check with `iio_info` or `iio_attr` what's there?

Ev1lbl0w, 2020-09-13:
The output of `iio_info` is:
```
$ Library version: 0.21 (git tag: v0.21)
Compiled with backends: local xml ip usb serial
IIO context created with local backend.
Backend version: 0.21 (git tag: v0.21)
Backend description string: Linux ricardo 5.8.5-arch1-1 #1 SMP PREEMPT Thu, 27 Aug 2020 18:53:02 +0000 x86_64
IIO context has 2 attributes:
	local,kernel: 5.8.5-arch1-1
	uri: local:
IIO context has 0 devices:
```

If I do `iio_info --auto` though, it finds the device:
```
$ iio_info --auto       
Library version: 0.21 (git tag: v0.21)
Compiled with backends: local xml ip usb serial
Multiple contexts found. Please select one using --uri:
	0: (Local IIO devices on HP) [local:]
	1: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104473541196000417000800f1ff1e227f [usb:1.10.5]
```

And then, by specifying the URI I can get device info thorugh `iio_info` and `iio_attr` (https://pastebin.com/YS6cXNc4)

zuckschwerdt, 2020-09-28:
Ok, I suspect that we try to open that other IIO context (what is that?) and obviously there are no Pluto-specific devices there. We should make sure that the context we are looking at actually is a Puto and also maybe try to find more contexts?

closed 2020-09-28 by zuckschwerdt

zuckschwerdt, 2020-09-28:
There is now a sanity check so we don't try to work with a non-PlutoSDR IIO context. You need to use a `uri` or `hostname` (see uri and network in https://analogdevicesinc.github.io/libiio/master/libiio/index.html#creating_context) arg to select the PlutoSDR context if the default IIO context isn't the right one.

commit 93717b32 (2020-09-28): Fix crash on non-PlutoSDR IIO context (closes #35)