pothosware/SoapyPlutoSDR #36: Link against minor libiio soname version

FFY00, 2020-09-13:
It looks like the libiio API can change in minor soname bumps, so we should be linking against the minor soname version. Currently libad9361 is linking against `libiio.so.0` but it should be linking against `libiio.so.21`.

zuckschwerdt, 2020-09-13:
0.21 seems to be the latest version. Are there API or ABI changes with current master?

guruofquality, 2020-09-13:
* https://github.com/pothosware/SoapyPlutoSDR/blob/master/FindLibIIO.cmake

The script for cmake above should be finding libiio.so which is also a symlink. 

* libiio.so should point to the ABI specific library symlink which is libiio.so.0
* then libiio.so.0 points to the specific actual library with version info libiio.so.0.21

If developers do it right, you wont see an ABI issue because of the symlinks, but many times the ABI changes and the number does not get bumped. I would guess that `libiio.so.0` has never seen an ABI increment. Therefore you have to recompile when the iio library is reinstalled.

FFY00, 2020-09-13:
Yes, that is exactly the issue. Because the libiio upstream does not provide proper ABI increments, I think soapyplutosdr should link to `libiio.so.0.21` so that we do not see random hard to track crashes.

In Arch Linux we are rebuilding on every libiio update.