pothosware/SoapyPlutoSDR #37: [ERROR] no device found in this context.?

eljuligallego, 2020-11-18:
Hi just install from source in the jetson nano but the command :

SoapySDRUtil--probe="driver=plutosdr,hostname=192.168.2.1" 

give me this error but the pluto is connected and detected with the iio_info -n 192.168.2.1 command.

Also SoapySDRUtil --find detects it:
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = plutosdr
  driver = plutosdr
  uri = local:

Found device 1
  device = plutosdr
  driver = plutosdr
  uri = usb:1.6.5

how should i run the command?.

LamaBleu, 2020-12-05:
Hi I'm facing the same issue.
I'm a regular user of SoapySDR/SoapyPlutoSDR on different platforms with no issue since 2 years.   
  

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = plutosdr
  driver = plutosdr
  uri = local:

Found device 1
  device = plutosdr
  driver = plutosdr
  uri = usb:1.12.5

```
  

```
SoapySDRUtil --probe="driver=plutosdr,uri=usb:1.12.5"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=plutosdr,uri=usb:1.12.5
[ERROR] no device found in this context.
Error probing device: no device found in this context

```
  
iio_info is OK.  

However this issue seems related to the platform : Jetson TX2, and Jetson Nano devkit.  
IIO is also present for sensors and I suspect the device 0 ( URI local:) detected by Soapy as pluto is the sensor.  

```
iio_info -s
Library version: 0.21 (git tag: 64f2e41)
Compiled with backends: local xml ip usb
Available contexts:
	0: (Local IIO devices on NVIDIA Jetson Nano 2GB Developer Kit) [local:]
	1: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104400b83991xxxxxxxxxxxxx [usb:1.12.5]

```

I compiled libiio from sources, and have no issue using GNUradio 3.8 and GQRX with PlutoSDR support.  
rtl_433 not working because it's depends on SoapyPlutoSDR.  

**Workaround :**  
Situation backs to normal re-compiling libiio without local backend : 
`cmake -DWITH_LOCAL_BACKEND=NO -DWITH_IIOD=NO ../`  

```
nvidia@NanoDev:~/libiio/build$ iio_info -s
Library version: 0.21 (git tag: 64f2e41)
Compiled with backends: xml ip usb serial
Unable to create Local IIO context : Function not implemented
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104400b83991xxxxxxxxxxxxx [usb:1.12.5]

```
Once libiio is reinstalled, SoapySDR is then working perfectly :  
```

nvidia@NanoDev:~/libiio/build$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = plutosdr
  driver = plutosdr
  uri = usb:1.12.5
```
  
```
nvidia@NanoDev:~/libiio/build$ SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9364
  ad9361-phy,xo_correction=39999898
  backend_version=0.21 (git tag: v0.21  )
  fw_version=v0.32-dirty
  hw_model=Analog Devices PlutoSDR Rev.B (Z7010-AD9364)

```
and rtl_433 is now working :  

```
nvidia@NanoDev:~/libiio/build$ rtl_433 -d driver=plutosdr
rtl_433 version 20.11-3-g095ab03 branch master at 202012041155 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/nvidia/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
[INFO] Using format CS16.
[INFO] Auto setting Buffer Size: 262144
[INFO] Set MTU Size: 262144
[NOTICE] sample rate needs a FIR setting loaded.
[INFO] Auto setting Buffer Size: 8192
[INFO] Set MTU Size: 8192
Sample rate set to 249999 S/s.
Tuner set to automatic gain.
Tuner gain set to Auto.
[INFO] Has direct RX copy: 1
Tuned to 433.920MHz.

```

Hope this can help.

eljuligallego, 2020-12-07:
Thanks so much!, now is working.

eljuligallego, 2020-12-07:
It's on the Jetson nano 2GB version...

closed 2020-12-15 by eljuligallego

robcazzaro, 2022-02-20:
I'm running Ubunto 20.4  on a Microsoft Surface 3, and getting this error, I think because of the IMU sensor present in the Surface

I tried 

```
~/SDR/SoapyPlutoSDRnew/build$ iio_info -s
Library version: 0.23 (git tag: 92d6a35)
Compiled with backends: local xml ip usb serial
Available contexts:
	0: (magn_3d, gyro_3d, dev_rotation, incli_3d, accel_3d, als on Microsoft Corporation) [local:]
	1: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=1044734c9605000df9ff0500deb64f2518 [usb:1.12.5]
	2: 192.168.2.1 (Analog Devices PlutoSDR Rev.C (Z7010-AD9364)), serial=1044734c9605000df9ff0500deb64f2518 [ip:pluto.local]

~/SDR/SoapyPlutoSDRnew/build$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g722207a2
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libPlutoSDRSupport.so (0.2.1)
Available factories... plutosdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

~/SDR/SoapyPlutoSDRnew/build$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = PlutoSDR
  driver = plutosdr
  label = PlutoSDR #0 local:
  uri = local:

Found device 1
  device = PlutoSDR
  driver = plutosdr
  label = PlutoSDR #0 usb:1.12.5
  uri = usb:1.12.5

Found device 2
  device = PlutoSDR
  driver = plutosdr
  label = PlutoSDR #0 ip:pluto.local
  uri = ip:pluto.local

~/SDR/SoapyPlutoSDRnew/build$ SoapySDRUtil --probe="driver=plutosdr,usb:1.12.5"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=plutosdr,usb:1.12.5
[INFO] Opening PlutoSDR #0 local:...
[ERROR] no device found in this context.
Error probing device: no device found in this context

~/SDR/SoapyPlutoSDRnew/build$ SoapySDRUtil --probe="driver=plutosdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=plutosdr
[INFO] Opening PlutoSDR #0 local:...
[ERROR] no device found in this context.
Error probing device: no device found in this context```

But compiling the suggested commit, seems to make no difference... anything I can try?

commit ceb63bad (2022-07-01): Add uri key to match devices, ensure that libiio context is a PlutoSDR (#42)

* Respect specified `uri` when searching for devices
* Ensure that discovered libiio context is a PlutoSDR

Fixes #37