pothosware/SoapyPlutoSDR #40: SoapyPlutoSDR included in Pothos?

androbi-com, 2021-04-29:
Please forgive me if this is a stupid question, but should SoapyPlutoSDR be included with the Pothos distribution? This [page](https://github.com/pothosware/SoapyPlutoSDR/wiki) made me think so, but my device is not detected. And, no "pluto" related file in lib/SoapySDR/modules0.8. I installed the latest release PothosSDR-2021.02.28-vc16-x64. SDR Console works on the same machine. Any hints for getting this to work?

zuckschwerdt, 2021-04-29:
It's there but as component: https://github.com/pothosware/PothosSDR/blob/master/SetupNSIS.cmake#L153
You need to enable "PlutoSDR SoapySDR binding (experimental)" (in the "experimental" group) while installing.

androbi-com, 2021-04-29:
Thanx a lot! Maybe a short note about the experimental status could be added to (https://github.com/pothosware/SoapyPlutoSDR/wiki)?

guruofquality, 2021-04-29:
Just for reference this is the reason:

* https://github.com/pothosware/SoapyPlutoSDR/issues/24
* https://github.com/analogdevicesinc/libiio/issues/586

I had to make it optional because it can mess with other libusb devices, and it was an ongoing support issue that kept coming up.

commit 28fdd204 (2021-05-01): Add PothosSDR note (closes #40)
commit ca40ea9a (2021-05-01): Add PothosSDR note (closes #40)
closed 2021-05-01 by zuckschwerdt