pothosware/SoapyPlutoSDR #47: Support for second RX channel?

nategri, 2021-09-13:
I know the internal AD chipset permits a second RX channel---how difficult would it be to add support for this second channel? I have a Pluto+ (Chinese variant) with dual receive, and I was hoping to use the second channel in a calibration context with SoapyPower.

Any hints you could give would be much appreciated!

zuckschwerdt, 2021-09-13:
Can you give a dump of `iio_info` (`iio_info --scan`)?

nategri, 2021-09-14:
Sure thing---and let me know if this isn't what you were looking for:

```
pi@raspberrypi:~ $ iio_info --network 192.168.2.1
Library version: 0.16 (git tag: v0.16)
Compiled with backends: local xml ip usb serial
IIO context created with network backend.
Backend version: 0.21 (git tag: v0.21  )
Backend description string: 192.168.2.1 Linux (none) 4.19.0-g8e9fc41dc7db #25 SMP PREEMPT Mon Sep 28 09:16:27 CST 2020 armv7l
IIO context has 9 attributes:
	hw_model: Analog Devices PlutoSDR Rev.C (Z7010-AD9364)
	hw_model_variant: 1
	hw_serial: 
	fw_version: v0.32-1-g7bdc-dirty
	ad9361-phy,xo_correction: 40000000
	ad9361-phy,model: ad9364
	local,kernel: 4.19.0-g8e9fc41dc7db
	uri: local:
	ip,ip-addr: 192.168.2.1
IIO context has 4 devices:
	iio:device0: ad9361-phy
		9 channels found:
			altvoltage0: RX_LO (output)
			8 channel-specific attributes found:
				attr  0: external value: 0
				attr  1: fastlock_load value: 0
				attr  2: fastlock_recall ERROR: Invalid argument (-22)
				attr  3: fastlock_save value: 0 149,63,10,139,235,227,171,227,227,227,10,235,227,227,79,227
				attr  4: fastlock_store value: 0
				attr  5: frequency value: 98499998
				attr  6: frequency_available value: [70000000 1 6000000000]
				attr  7: powerdown value: 0
			altvoltage1: TX_LO (output)
			8 channel-specific attributes found:
				attr  0: external value: 0
				attr  1: fastlock_load value: 0
				attr  2: fastlock_recall ERROR: Invalid argument (-22)
				attr  3: fastlock_save value: 0 64,208,88,223,217,81,223,223,223,223,223,223,223,163,223,217
				attr  4: fastlock_store value: 0
				attr  5: frequency value: 2450000000
				attr  6: frequency_available value: [46875001 1 6000000000]
				attr  7: powerdown value: 0
			out:  (input)
			1 channel-specific attributes found:
				attr  0: voltage_filter_fir_en value: 1
			temp0:  (input)
			1 channel-specific attributes found:
				attr  0: input value: 42982
			voltage0:  (output)
			10 channel-specific attributes found:
				attr  0: filter_fir_en value: 1
				attr  1: hardwaregain value: -10.000000 dB
				attr  2: hardwaregain_available value: [-89.750000 0.250000 0.000000]
				attr  3: rf_bandwidth value: 18000000
				attr  4: rf_bandwidth_available value: [200000 1 40000000]
				attr  5: rf_port_select value: A
				attr  6: rf_port_select_available value: A B
				attr  7: rssi value: 0.00 dB
				attr  8: sampling_frequency value: 3000000
				attr  9: sampling_frequency_available value: [520833 1 30720000]
			voltage0:  (input)
			15 channel-specific attributes found:
				attr  0: bb_dc_offset_tracking_en value: 1
				attr  1: filter_fir_en value: 1
				attr  2: gain_control_mode value: slow_attack
				attr  3: gain_control_mode_available value: manual fast_attack slow_attack hybrid
				attr  4: hardwaregain value: 73.000000 dB
				attr  5: hardwaregain_available value: [-1 1 73]
				attr  6: quadrature_tracking_en value: 1
				attr  7: rf_bandwidth value: 18000000
[... 342 lines cut ...]
			6 channel-specific attributes found:
				attr  0: calibbias value: 0
				attr  1: calibphase value: 0.000000
				attr  2: calibscale value: 1.000000
				attr  3: samples_pps ERROR: No such device (-19)
				attr  4: sampling_frequency value: 3000000
				attr  5: sampling_frequency_available value: 3000000 375000 
			voltage1:  (input, index: 1, format: le:S12/16>>0)
			6 channel-specific attributes found:
				attr  0: calibbias value: 0
				attr  1: calibphase value: 0.000000
				attr  2: calibscale value: 1.000000
				attr  3: samples_pps ERROR: No such device (-19)
				attr  4: sampling_frequency value: 3000000
				attr  5: sampling_frequency_available value: 3000000 375000 
		3 buffer-specific attributes found:
				attr  0: data_available value: 0
				attr  1: length_align_bytes value: 8
				attr  2: watermark value: 2048
		2 debug attributes found:
				debug attr  0: pseudorandom_err_check value: CH0 : PN9 : Out of Sync : PN Error
CH1 : PN9 : Out of Sync : PN Error
				debug attr  1: direct_reg_access value: 0x0
pi@raspberrypi:~ $ 
```

zuckschwerdt, 2021-09-14:
Look at `ad9361-phy`: there are limited controls on the second channel, i.e. no gain_control_mode, hardwaregain, rf_port_select values on voltage2 (RX2/TX2). I guess those are all linked to voltage0 (RX1/TX1)? Useful for MIMO only then?

Generally there is support for multiple channels, if you set https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Settings.cpp#L118 to "2" and recompile you should be able to stream both RX or both TX at the same time. E.g. `setupStream()` does recognize `channels={0,1}`.

zuckschwerdt, 2021-09-14:
Interestingly https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#tx_attenuation_control talks about "attenuation/gain can be individually controlled for TX1 and TX2" at "voltage1 hardwaregain", but there is no voltage1 here, I assume it's the voltage2 channel, and there is no hardwaregain value.

nategri, 2021-09-14:
Ah! So you're saying the RX2 channel just has whatever gain the RX1 one has?

Also I did *exactly* that one line code change and recompile (1->2) before I opened the issue. I thought what I was trying was too naive because it looked like the data coming from both channels was the same. But maybe I got my observation wrong! Will look again.

zuckschwerdt, 2021-09-14:
Yes, and most of the other settings too. IIO is the API. any value you don't find there is not an API setting.

moemuses, 2022-05-30:
Hi @zuckschwerdt 
I have pluto+ with `SoapySDRUtil --probe="driver=plutosdr"`
we have only one RX channel "numbered  0" : 
`----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: A_BALANCED
  Full gain range: [0, 73] dB
    PGA gain range: [0, 73] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.260417, 61.44] MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz
`
but inside pluto device inside `cd /sys/bus/iio/devices/iio:device0` we heve :
`in_voltage0_gain_control_mode        
in_voltage0_hardwaregain             
in_voltage0_hardwaregain_available   
in_voltage0_rf_port_select           
in_voltage0_rssi                     
in_voltage1_gain_control_mode        
in_voltage1_hardwaregain             
in_voltage1_hardwaregain_available   
in_voltage1_rf_port_select           
in_voltage1_rssi                     
`
both RX port have their own parameters.

My Question, can i use the second channel RX2 using SoapyPlutoSDR ?

zuckschwerdt, 2022-05-30:
The AD9363 is a 2x2 transceiver, but I wasn't aware that the second channel is connected in any hardware version.
That "Pluto+" seems to be a no-name knock-off with some extra features (2nd channels, Gigabit ethernet).

You can change "getNumChannels" like written above to test things. But we are not explicitly using the "voltage1".

Can you give a dump of `iio_info` (`iio_info --scan`)?

moemuses, 2022-05-31:
```
Library version: 0.21 (git tag: v0.21)
Compiled with backends: local xml ip usb serial
Using auto-detected IIO context at URI "usb:1.70.5"
IIO context created with usb backend.
Backend version: 0.21 (git tag: v0.21  )
Backend description string: Linux (none) 5.4.0-gb05d16429dac-dirty #4 SMP PREEMPT Fri Apr 23 23:55:52 CST 2021 armv7l
IIO context has 15 attributes:
	hw_model: Analog Devices PlutoSDR Rev.C (Z7010-AD9363A)
	hw_model_variant: 1
	hw_serial: 104000d6500700130d002000e1ced35cae
	fw_version: v0.33-3-gd382-dirty
	ad9361-phy,xo_correction: 40000000
	ad9361-phy,model: ad9363a
	local,kernel: 5.4.0-gb05d16429dac-dirty
	uri: usb:1.70.5
	usb,idVendor: 0456
	usb,idProduct: b673
	usb,release: 2.0
	usb,vendor: Analog Devices Inc.
	usb,product: PlutoSDR (ADALM-PLUTO)
	usb,serial: 104000d6500700130d002000e1ced35cae
	usb,libusb: 1.0.24.11584
IIO context has 4 devices:
	iio:device0: ad9361-phy
		11 channels found:
			altvoltage1: TX_LO (output)
			8 channel-specific attributes found:
				attr  0: external value: 0
				attr  1: fastlock_load value: 0
				attr  2: fastlock_recall ERROR: Invalid argument (-22)
				attr  3: fastlock_save value: 0 242,254,255,62,127,255,63,251,251,255,124,86,254,86,111,84
				attr  4: fastlock_store value: 0
				attr  5: frequency value: 2450000000
				attr  6: frequency_available value: [325000000 1 3800000000]
				attr  7: powerdown value: 0
			voltage1:  (output)
			10 channel-specific attributes found:
				attr  0: filter_fir_en value: 0
				attr  1: hardwaregain value: -10.000000 dB
				attr  2: hardwaregain_available value: [-89.750000 0.250000 0.000000]
				attr  3: rf_bandwidth value: 18000000
				attr  4: rf_bandwidth_available value: [200000 1 40000000]
				attr  5: rf_port_select value: A
				attr  6: rf_port_select_available value: A B
				attr  7: rssi value: 0.00 dB
				attr  8: sampling_frequency value: 30720000
				attr  9: sampling_frequency_available value: [2083333 1 30720000]
			voltage0:  (input)
			15 channel-specific attributes found:
				attr  0: bb_dc_offset_tracking_en value: 1
				attr  1: filter_fir_en value: 0
				attr  2: gain_control_mode value: slow_attack
				attr  3: gain_control_mode_available value: manual fast_attack slow_attack hybrid
				attr  4: hardwaregain value: 71.000000 dB
				attr  5: hardwaregain_available value: [-3 1 71]
				attr  6: quadrature_tracking_en value: 1
				attr  7: rf_bandwidth value: 18000000
				attr  8: rf_bandwidth_available value: [200000 1 56000000]
				attr  9: rf_dc_offset_tracking_en value: 1
				attr 10: rf_port_select value: A_BALANCED
				attr 11: rf_port_select_available value: A_BALANCED B_BALANCED C_BALANCED A_N A_P B_N B_P C_N C_P TX_MONITOR1 TX_MONITOR2 TX_MONITOR1_2
				attr 12: rssi value: 96.25 dB
				attr 13: sampling_frequency value: 30720000
				attr 14: sampling_frequency_available value: [2083333 1 30720000]
			voltage1:  (input)
			15 channel-specific attributes found:
				attr  0: bb_dc_offset_tracking_en value: 1
				attr  1: filter_fir_en value: 0
				attr  2: gain_control_mode value: slow_attack
[... 511 lines cut ...]
				attr  1: calibphase value: 0.000000
				attr  2: calibscale value: 1.000000
				attr  3: samples_pps ERROR: No such device (-19)
				attr  4: sampling_frequency value: 30720000
				attr  5: sampling_frequency_available value: 30720000 3840000 
			voltage3:  (input, index: 3, format: le:S12/16>>0)
			6 channel-specific attributes found:
				attr  0: calibbias value: 0
				attr  1: calibphase value: 0.000000
				attr  2: calibscale value: 1.000000
				attr  3: samples_pps ERROR: No such device (-19)
				attr  4: sampling_frequency value: 30720000
				attr  5: sampling_frequency_available value: 30720000 3840000 
		3 buffer-specific attributes found:
				attr  0: data_available value: 0
				attr  1: length_align_bytes value: 8
				attr  2: watermark value: 2048
		2 debug attributes found:
				debug attr  0: pseudorandom_err_check value: CH0 : PN9 : Out of Sync : PN Error
CH1 : PN9 : Out of Sync : PN Error
CH2 : PN9 : Out of Sync : PN Error
CH3 : PN9 : Out of Sync : PN Error
				debug attr  1: direct_reg_access value: 0x0
		No trigger on this device
```

zuckschwerdt, 2022-05-31:
Thanks. Interesting, so that really is a AD9363 (2 Rx, 2 Tx) and not the AD9364 (1x1) we otherwise see.
We should be able to detect the presence of the `voltage1` in/out channel and then offer the second channel automatically.

I'll try to soon get a basic PR together. I'm not likely going to buy that hardware, so you'll have to help and test several version until this works ;)

moemuses, 2022-05-31:
I see in [sdrangle](https://github.com/f4exb/sdrangel/blob/master/plugins/samplemimo/plutosdrmimo/readme.md) they have multiple channel. it's working for me.
but the plugin is not complete and have a lot of bugs. also we can't receive IQ data from both channel at same time.

i'm thankfully and ready to test any code until it works.

zuckschwerdt, 2022-06-12:
The branch https://github.com/pothosware/SoapyPlutoSDR/tree/feat-multichn should work now to select the second channel.
I have not checked or implemented anything for both channels at the same time yet.

zuckschwerdt, 2022-06-28:
@moemuses did you have a chance to test the changes from branch feat-multichn?

moemuses, 2022-06-29:
@zuckschwerdt, yes i'm ready. just tell be how exactly i can test it ?

zuckschwerdt, 2022-06-29:
Do `git clone https://github.com/pothosware/SoapyPlutoSDR.git`
or change to an existing SoapyPlutoSDR folder and `git fetch`
then change the branch `git checkout feat-multichn`
now build and install as usual `mkdir build ; cd build ; cmake .. && make && make install`

moemuses, 2022-06-29:
yes i mean what after make the project ? how can i test the second channel ?

moemuses, 2022-06-29:
perform `SoapySDRUtil --probe="driver=plutosdr"`
having 2 TX and 2 RX

```
----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 2 Rx, 2 Tx
  Timestamps: NO
  Sensors: xadc_temp0, xadc_voltage0, xadc_voltage1, xadc_voltage2, xadc_voltage3, xadc_voltage4, xadc_voltage5, xadc_voltage6, xadc_voltage7, xadc_voltage8, adm1177_current0, adm1177_voltage0, ad9361-phy_temp0, ad9361-phy_voltage2

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: A_BALANCED, B_BALANCED, C_BALANCED, A_N, A_P, B_N, B_P, C_N, C_P, TX_MONITOR1, TX_MONITOR2, TX_MONITOR1_2
  Full gain range: [0, 73] dB
    PGA gain range: [0, 73] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.0651042, 61.44] MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz

----------------------------------------------------
-- RX Channel 1
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: A_BALANCED, B_BALANCED, C_BALANCED, A_N, A_P, B_N, B_P, C_N, C_P, TX_MONITOR1, TX_MONITOR2, TX_MONITOR1_2
  Full gain range: [0, 73] dB
    PGA gain range: [0, 73] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.0651042, 61.44] MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=32768]
  Antennas: A, B
  Full gain range: [0, 89] dB
    PGA gain range: [0, 89] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.0651042, 61.44] MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz

----------------------------------------------------
-- TX Channel 1
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=32768]
  Antennas: A, B
  Full gain range: [0, 89] dB
    PGA gain range: [0, 89] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.0651042, 61.44] MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz

```

also when perform `iio_readdev -u ip:192.168.2.1 -b 32768 -s 0 cf-ad9361-lpc | pv > /dev/null` after change sample rate to 2.1Mhz, the normal Pluto make [8.01MiB/s] of data and Pluto+ have [16.0MiB/s]. maybe one IQ pair from RX0 and another IQ pair from RX1

moemuses, 2022-06-29:
the 16MiB/s come from iio library not Soapy library.
the only UI application i can fine with dual channel receiving is sdrangel.

and the waterfall changes as i change the RX port.
but also sdrangel work directly with iio not Soapy

zuckschwerdt, 2022-06-29:
https://github.com/BatchDrake/SigDigger uses Soapy and allows to select the channel.

moemuses, 2022-06-29:
in Sigdigger, regardless of using #channel number, alwayes same waterfall and data

moemuses, 2022-06-29:
But in Device name, only appear plutosdr usb:x.x.x. nothing refer to Soapy

zuckschwerdt, 2022-06-30:
I half expected that ;) But now we have test code I'll go on to add multi channel support. In a day or so.

fpw, 2022-07-11:
FYI, the latest hardware revisions of the original Pluto are also 2x2 now. The two new channels don't have an external connector, but there are internal connectors that can easily be connected: https://www.youtube.com/watch?v=ph0Kv4SgSuI

So this doesn't only apply to "Pluto+" anymore.

zuckschwerdt, 2022-07-11:
Thanks! I need to go through the different channel copy cases mentioned above and add multi channel. And then we likely need separate rate/freq per channel.

AlexeyBarsuk, 2024-05-04:
Any news on multi channel support ? 
There are many SDRs for today, based on ad9361/9363 with 2R2T mode. They are original Pluto, Chinese Pluto+, Chinese AntSDR. It would be great to add good multi channel support.
As for me I used branch feat-multichn, however I can not make it work with new iio versions.

zuckschwerdt, 2024-05-05:
I think the basics are there with the feat-multichn branch, but there are quiet some details around dependent settings and such. I only have a Rev.B and don't think I can finish this without the hardware to test.