pothosware/SoapyPlutoSDR #52: Revert libusb check for known ADALM-Pluto VID:PID

zuckschwerdt, 2022-07-01:
We should revert the libusb check for known ADALM-Pluto VID:PID from #50 once libiio 0.24 can be assumed the required minimum version.

Todo item for 2024 perhaps.

moemuses, 2022-07-02:
cmake error in last branch, and i can't return to last working code

```
undefined reference to `libusb_init'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x10c5): undefined reference to `libusb_get_device_list'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x110f): undefined reference to `libusb_get_device_descriptor'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x1149): undefined reference to `libusb_free_device_list'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x33b7): undefined reference to `libusb_free_device_list'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/PlutoSDRSupport.dir/build.make:151: libPlutoSDRSupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:123: CMakeFiles/PlutoSDRSupport.dir/all] Error 2

```

zuckschwerdt, 2022-07-02:
Typo fixed with 04943ed. If the error persists please open a new issue and report the `cmake` output.

moemuses, 2022-07-02:
thanks, it's working now, dont't forget to change it on multichannel branch also please.

alphafox02, 2022-07-10:
@zuckschwerdt I can't explain why this is happening at the moment, but now over on my laptop (not the Rock Pi X w/ sensor detected by libiio) I have no updated the Soapy Pluto and it seems to be causing an issue.

I am unable to see both the PlutoSDR and AntSDR via USB with SoapySDRUtil --find. However, if I revert to the most recent 8 days ago or so commit, rebuild and install, I can then see both the AntSDR and the PlutoSDR. 

This is strange, because I can almost say for sure that on the Rock Pi X I did not have this issue. I'm going to go commit by commit t forward ill it stops working again.

alphafox02, 2022-07-10:
Figured it out, my laptop had an older libiio. I did a get pull of the latest, rebuilt, and all is fine now with even your latest changes to Soapy for Pluto - which was likely how I was testing it on the Rock Pi X.