pothosware/SoapyPlutoSDR #53: Listed sample rates and bandwidths

catkira, 2022-07-10:
The driver lists these samplerates (and same for bandwidths)
```c
std::vector<double> SoapyPlutoSDR::listSampleRates( const int direction, const size_t channel ) const
{
	std::vector<double> options;
	options.push_back(65105);//25M/48/8+1
	options.push_back(1e6);
	options.push_back(2e6);
	options.push_back(3e6);
	options.push_back(4e6);
	options.push_back(5e6);
	options.push_back(6e6);
	options.push_back(7e6);
	options.push_back(8e6);
	options.push_back(9e6);
	options.push_back(10e6);
	return(options);
}
```
That results in being able to only select one of these sample rates in certain apps like SigDigger. However the PlutoSDR hardware can do custom sample rates like 1.5 MSPS and also higher sample rates like 15 MSPS. Because of the USB 2.0 that Pluto has, higher sample rates dont make much sense, but ANTSDR (a pluto clone with gbit ethernet) can stream 14-15 MSPS.
I wonder why the Soapy driver has to list specific sample rates? What if the device supports any sample rate in a given range? Can the driver signal that any sample rate in a range is supported?

zuckschwerdt, 2022-07-10:
The replacement API for that is [`getSampleRateRange()`](https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Device.hpp#L894-L909) and is present: https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Settings.cpp#L647-L663

Software supporting arbitrary rates should use that API.

catkira, 2022-07-10:
cool thanks!

closed 2022-07-10 by catkira

zuckschwerdt, 2022-07-10:
Also note that Soapy can be used on the Pluto itself, higher sample rates then are not a problem.

catkira, 2022-07-10:
What is the advantage of running Soapy on Pluto? Since Pluto does not have a Display, it only makes sense to run Soapy as a server with Soapy remote I guess. But the libiio network backend basically does the same. And running Soapy on Pluto does not save any network/usb bandwidth.
Why can Soapy do higher sample rates when running on Pluto? The USB 2.0 bottleneck is still there. Or do You mean to do some processing in Soapy and not transfer the raw data stream over network/usb ?

zuckschwerdt, 2022-07-10:
Why not do it, if we can hack that? But yes, processing. You can demodulate and decimate the band.
E.g. I use the Pluto stand-alone with just a usb stick for storage and one for wifi to run rtl_433 on the go.

catkira, 2022-07-10:
You run rtl_433 on Your laptop or on pluto? What do You store on USB, logfiles or output of rtl_433 or raw samples that can be analyzed later with rtl_433?

zuckschwerdt, 2022-07-10:
There is no laptop. But I can connect (wifi) from a phone to the Pluto to start/stop things or change modes. Log files, samples, raw dumps — it all works, the Pluto is a Linux machine like a Raspi after all ;)