pothosware/SoapyPlutoSDR #56: Segmentation fault while trying to capture signal with GR-GSM and PlutoSDR

VadimIzmalkov, 2023-02-24:
Hi everyone!

When I try to use PlutoSDR with GR-GSM to cature a signal get the error:

```CLI
$ grgsm_capture -a32 test.cfile -T10
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp hackrf bladerf rfspace soapy redpitaya
[INFO] Opening PlutoSDR #0 usb:1.8.5...
[INFO] Using format CF32.
Segmentation fault
```

while the RTL SDR works without problems:

```CLI
grgsm_capture -a32 test.cfile -T10
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp hackrf bladerf rfspace soapy redpitaya
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
rtlsdr_read_async returned with -5
```

I use:
- ubuntu 22.04 (Widows 10 + WSL)
- GNURadio 3.10.1.1 (Python 3.10.6)
- GR-GSM from https://github.com/bkerler/gr-gsm (maint-3.10_with_multiarfcn branch)

```CLI
$ SoapySDRUtil --info

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g9c4fa324
API Version: v0.8.200
ABI Version: v0.8-2
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8-2
Module found: /usr/local/lib/SoapySDR/modules0.8-2/libPlutoSDRSupport.so (0.2.1-a07c372)
Available factories... plutosdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

Any ideas?

zuckschwerdt, 2023-02-24:
Make an ASAN build of the executeable to catch the bug maybe.
Extend CFLAGS with "-ggdb -fsanitize=undefined -fsanitize=address -fno-omit-frame-pointer"

Or run in `gdb` and get a stacktrace. Build with "-g" at least to get the symbols.

VadimIzmalkov, 2023-02-25:
Please confirm that I am doing everything right

```CLI
cd SoapyPlutoSDR
mkdir build
cd build
cmake -E env "CXXFLAGS=-ggdb -fsanitize=undefined -fsanitize=address -fno-omit-frame-pointer" cmake ..
make
sudo make install
```

If it is right, the output is

```CLI
$ grgsm_capture -a32 test.cfile -T10
==1703==ASan runtime does not come first in initial library list; you should either link runtime to your application or manually preload it with LD_PRELOAD.
```

zuckschwerdt, 2023-02-25:
You should build your executable with -fsanitize=address flag too.

To get a feel for ASAN compile and run this test code
```cpp
#include <iostream>
using namespace std;
int main()
{
	int* p = nullptr;
	cout << "p = " << *p << '\n';
}
```
with
```sh
g++ test.cpp -fsanitize=address && ./a.out
```

zuckschwerdt, 2023-07-24:
This might have been fixed by #57.