pothosware/SoapyPlutoSDR #60: SoapyPluto build issue

nmaster2042, 2023-12-25:
I built last  libiio (v1.0) and libad9361 (v0.2).

Now I got errors liking SoapyPlutoSDR

```
[100%] Linking CXX shared module libPlutoSDRSupport.so
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Registration.cpp.o: in function `find_PlutoSDR(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)':
PlutoSDR_Registration.cpp:(.text+0x11ab): undefined reference to `iio_create_scan_context'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x11d5): undefined reference to `iio_scan_context_get_info_list'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x132c): undefined reference to `iio_context_info_list_free'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x1336): undefined reference to `iio_scan_context_destroy'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x143c): undefined reference to `iio_create_network_context'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x2045): undefined reference to `iio_context_info_get_uri'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x204d): undefined reference to `iio_create_context_from_uri'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x206a): undefined reference to `iio_context_info_get_uri'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x22d5): undefined reference to `iio_context_info_list_free'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x22df): undefined reference to `iio_scan_context_destroy'
/usr/bin/ld: PlutoSDR_Registration.cpp:(.text+0x2fe7): undefined reference to `iio_context_info_list_free'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::setAntenna(int, unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
PlutoSDR_Settings.cpp:(.text+0x1ae): undefined reference to `iio_channel_attr_write'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x1f3): undefined reference to `iio_channel_attr_write'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::setGainMode(int, unsigned long, bool)':
PlutoSDR_Settings.cpp:(.text+0x272): undefined reference to `iio_channel_attr_write'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x297): undefined reference to `iio_channel_attr_write'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::setGain(int, unsigned long, double)':
PlutoSDR_Settings.cpp:(.text+0x307): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x358): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::getGain(int, unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
PlutoSDR_Settings.cpp:(.text+0x3dd): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x43a): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::setFrequency(int, unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)':
PlutoSDR_Settings.cpp:(.text+0x4c3): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x513): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::getFrequency(int, unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
PlutoSDR_Settings.cpp:(.text+0x59d): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x5fd): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::getSampleRate(int, unsigned long) const':
PlutoSDR_Settings.cpp:(.text+0x68d): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x6ea): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::setBandwidth(int, unsigned long, double)':
PlutoSDR_Settings.cpp:(.text+0x763): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x7b0): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::getBandwidth(int, unsigned long) const':
PlutoSDR_Settings.cpp:(.text+0x82d): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x88a): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::setSampleRate(int, unsigned long, double)':
PlutoSDR_Settings.cpp:(.text+0x953): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x987): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0xa30): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0xa5d): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::~SoapyPlutoSDR()':
PlutoSDR_Settings.cpp:(.text+0xcea): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0xd10): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0xd43): undefined reference to `iio_channel_attr_read_longlong'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0xd6c): undefined reference to `iio_channel_attr_write_longlong'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::get_sensor_value(iio_channel*) const':
PlutoSDR_Settings.cpp:(.text+0x137b): undefined reference to `iio_channel_attr_read'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x13c3): undefined reference to `iio_channel_attr_read'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x1401): undefined reference to `iio_channel_attr_read'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x1448): undefined reference to `iio_channel_attr_read'
/usr/bin/ld: CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Settings.cpp.o: in function `SoapyPlutoSDR::SoapyPlutoSDR(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)':
PlutoSDR_Settings.cpp:(.text+0x1b25): undefined reference to `iio_channel_attr_write'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x1ba0): undefined reference to `iio_channel_attr_write'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x1df1): undefined reference to `iio_create_context_from_uri'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x1ef7): undefined reference to `iio_create_network_context'
/usr/bin/ld: PlutoSDR_Settings.cpp:(.text+0x1f01): undefined reference to `iio_create_default_context'
[... cut ...] t&)':
PlutoSDR_Streaming.cpp:(.text+0x2212): undefined reference to `iio_channel_attr_write_bool'
/usr/bin/ld: PlutoSDR_Streaming.cpp:(.text+0x2332): undefined reference to `iio_channel_attr_write_bool'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/PlutoSDRSupport.dir/build.make:152: libPlutoSDRSupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:123: CMakeFiles/PlutoSDRSupport.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

Any issue with these lib versions ?

zuckschwerdt, 2023-12-25:
`libiio` is at `0.25` (there is no 1.0) and the errors look like the lib is not present.

nmaster2042, 2023-12-25:
OK, I understand my mistake:

Here is what we can read on the gihub of libiio:

> Since August 22th 2023, the "master" branch of libiio contains what will eventually become libiio v1.0. It features a brand new API, which is incompatible with libiio v0.25 and older. Have a look at [the wiki](https://github.com/analogdevicesinc/libiio/wiki/libiio_0_to_1) for a description of the API changes.
> 
> The old v0.x API can still be found in the [libiio-v0](https://github.com/analogdevicesinc/libiio/tree/libiio-v0) branch. Libiio v0.x is now considered legacy, and as such, only important bug fixes will be accepted into this branch.
> 
> Old programs compiled against libiio v0.x will still be able to run with libiio v1.0 and newer, as it provides a compatibility layer.

So there is a new V1 but seems to be very early stage for now.

zuckschwerdt, 2023-12-25:
Interesting! Let's keep this open as tracker until there is a stable libiio 1.0 to upgrade to.

gariac, 2025-02-26:
This problem is still present in 2025

`
/usr/bin/ld: PlutoSDR_Streaming.cpp:(.text+0x22ea): undefined reference to `iio_channel_attr_write_bool'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/PlutoSDRSupport.dir/build.make:153: libPlutoSDRSupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:88: CMakeFiles/PlutoSDRSupport.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

`

I did fresh builds of libiio and soapysdr

Is this something ADI needs to fix?

zuckschwerdt, 2025-02-26:
Did you check that you have the correct (i.e. from the 0.x line) libiio version installed?

gariac, 2025-03-05:
I have the latest version. Can I load two versions?
`
iio_info
iio_info version: 1.0 (git tag:2c5c1f8a)
Libiio version: 1.0 (git tag: 2c5c1f8a) backends: local ip usb xml
IIO context created with local backend.
Backend version: 1.0 (git tag: 2c5c1f8a)
Backend description string: Linux framework13 6.12.12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.12-1 (2025-02-02) x86_64
IIO context has 2 attributes:
        attr  0: local,kernel value: 6.12.12-amd64
        attr  1: uri value: local:
`

zuckschwerdt, 2025-03-05:
Currently you need a `0.x` version while compiling and when running. The latest version is still 0.26. https://github.com/analogdevicesinc/libiio/releases

gariac, 2025-03-06:
What is the downside to running an old version?

zuckschwerdt, 2025-03-06:
What is the upside of running an unreleased preview version?

gariac, 2025-03-16:
"Since August 22th 2023, the "main" branch of libiio contains what will eventually become libiio v1.0."
"Libiio v0.x is now considered legacy, and as such, only important bug fixes will be accepted into this branch."

ADI sounds like a politician  here. I thought 1.0 was the version to use, deeming the 0 rev legacy.  But rereading this, I suppose old is current. It isn't like going back and forth is a great effort. So I will run the old er proper rev and see what happens.

zuckschwerdt, 2025-03-16:
I think the best policy for broad compatibility is to track Debian, when they change libiio to 1.0 we'll change too.
If someone wants to prepare the needed changes as PR that will be very welcome. Adventurous users could then run and test that PR until the final switch.

git2fa, 2026-11-01:
Normally Debian was always used for the Desktop PC.
But after Version 11 (bullseye) it got less stable and in Version 13 wine did not work any more.
That's why Linux Mint has been choosen now, because everything works so far ...