pothosware/SoapyPlutoSDR #63: Plutosdr Transmits only once

Mas313, 2024-05-11:
Hi All,
I have been trying to send IQ samples, stored in a buffer, multiple times through Plutosdr but the SDR transmits only once, for sending multiple times the executable file needs to be re run.  But then again the plutoSDR transmits only once. 

My environment is UBUNTU and installed SoapyPlutoSDR with all dependies succesfully, i.e.  libiio-v0 (new version cannot be installed, gives error) ,    libad9361 & SoapySDR and using SoapySDR C API.

I have verified very first transmission by looking into spectrum using SDRsharp on another PC.
Strangely, the SoapySDRDevice_writeStream() returns correct number of bytes every time it is called but transmission is done by SDR only once. 

(txStream == NULL)
    	{
        	printf("setupStream fail: %s\n", SoapySDRDevice_lastError());
        	SoapySDRDevice_unmake(sdr);
        	return EXIT_FAILURE;
    	}
	printf(" After: SoapySDRDevice_setupStream \n");  
  
	SoapySDRDevice_activateStream(sdr, txStream, 0, 0, 0); //start streaming
	

After every transmission i also wait for key-press to continue to avoid any timing issue of PlutoSDR, if any , for transmission but no luck.
I am setting gain of PLutosdr manually
if ( SoapySDRDevice_setGain(sdr, SOAPY_SDR_TX, 0, 25.0)==0 )
   	{	
		printf("-----------------------Manual Gain SET \n"); 
   	}
   	else
	{
		printf("Gain Set Error\n");
		exit(0); 
	}

Initially, i though that might be somehow the Tx gain is reset to 0 after very first causing transmission with 0 power since writeStream function returns number of bytes sent and not 0, so i called SoapySDRDevice_setGain() after every transmission but no luck. But again its strange that when executeable is re-run than plutosdr transmits but for very first time only.
I transmit by filling buf array with IQ samples in a loop

        void *buffs[] = {buf}; //array of buffers
        int flags=0; //flags set by receive operation
        long long timeNs=1000000; //timestamp for tx buffer
        ret= SoapySDRDevice_writeStream(sdr, txStream, &buffs[0], j, &flags, &timeNs, 100000); //100000 

Any solution please.

zuckschwerdt, 2024-05-11:
I've just tested and confirmed that this case (continue transmission after some pause) works well, with or without SoapySDRDevice_activateStream()+SoapySDRDevice_deactivateStream(). (using slightly hacked `soapy_transmit()` in https://github.com/triq-org/tx_tools)

Could there be something amiss in your code?

Mas313, 2024-05-11:
This is my code, i am not able to send multiple times, suggest what i need to include

#include <stdio.h> //printf
#include <stdlib.h> //free
#include <complex.h>
#include<math.h>
#include <stdint.h> //for 16bit int
#include <string.h>
#include <SoapySDR/Device.h> 
#include <SoapySDR/Formats.h>

#define M_PI	3.14159

#define PI 3.14

#define SHRT_MAX	3276

FILE	*fptr;

char *filename;

int main(int argc,char *argv[])
{

////////////////////////////////////// Soapy Init Begins ////////////////////////////////////////////////////////////

size_t length;
    //enumerate devices
 printf("Before Enumerate \n");
    SoapySDRKwargs *results = SoapySDRDevice_enumerate(NULL, &length);
 printf("After Enumerate \n");

printf("Before Found Device Loop \n");
    for (size_t i = 0; i < length; i++)
    {
        printf("Found device #%d: ", (int)i);
        for (size_t j = 0; j < results[i].size; j++)
        {
            printf("%s=%s, ", results[i].keys[j], results[i].vals[j]);
        }
        printf("\n");
    }
printf("After Found Device Loop \n");

printf("Before SoapySDRKwargsList_clear \n");
    SoapySDRKwargsList_clear(results, length);
printf("After SoapySDRKwargsList_clear \n");

    //create device instance
    //args can be user defined or from the enumeration result
    SoapySDRKwargs args = {};
    SoapySDRKwargs_set(&args, "driver", "plutosdr");
    SoapySDRDevice *sdr = SoapySDRDevice_make(&args);
    SoapySDRKwargs_clear(&args);

printf("After SoapySDRKwargs_clear(&args) \n");
    if (sdr == NULL)
    {
        printf("Errorr SoapySDRDevice_make fail: %s\n", SoapySDRDevice_lastError());
        return EXIT_FAILURE;
    }
printf("After SoapySDRDevice_make fail \n");

//query device info
    char** names = SoapySDRDevice_listAntennas(sdr, SOAPY_SDR_TX, 0, &length);
    printf("Tx antennas: ");
    for (size_t i = 0; i < length; i++) printf("%s, ", names[i]);
    printf("\n");
    SoapySDRStrings_clear(&names, length);

printf(" Before Applying Setting for Sample Rate \n");
    if (SoapySDRDevice_setSampleRate(sdr, SOAPY_SDR_TX, 0, 4e6) != 0)
    {
        printf("setSampleRate fail: %s\n", SoapySDRDevice_lastError());
    }
printf("After Applying Setting for Sample Rate \n");

printf(" Before: SoapySDRDevice_setFrequency \n");	  //912.3e6
    if (SoapySDRDevice_setFrequency(sdr, SOAPY_SDR_TX, 0, 1100e6, NULL) != 0)
    {
        printf("setFrequency fail: %s\n", SoapySDRDevice_lastError());
    }
printf(" After: SoapySDRDevice_setFrequency \n");
    //setup a stream (complex floats)

 if ( SoapySDRDevice_setGain(sdr, SOAPY_SDR_TX, 0, 25.0)==0 )
   {	
	printf("-----------------------Manual Gain SET \n"); 
   }
   else
	{
	printf("Gain Set Error\n");
	exit(0); 
	}

printf(" Before: SoapySDRDevice_setupStream \n");			// float 32 bit SOAPY_SDR_F32
    //SoapySDRStream *txStream = SoapySDRDevice_setupStream(sdr, SOAPY_SDR_TX, SOAPY_SDR_CF32, NULL, 0, NULL); //SOAPY_SDR_CF32
  
SoapySDRStream *txStream = SoapySDRDevice_setupStream(sdr, SOAPY_SDR_TX, SOAPY_SDR_CF32, NULL, 0, NULL);
	    if (txStream == NULL)
    		{
        	printf("setupStream fail: %s\n", SoapySDRDevice_lastError());
        	SoapySDRDevice_unmake(sdr);
        	return EXIT_FAILURE;
    		}
		printf(" After: SoapySDRDevice_setupStream \n");  
  
		SoapySDRDevice_activateStream(sdr, txStream, 0, 0, 0); //start streaming
		printf(" After calling--------  SoapySDRDevice_activateStream   \n");
		       

//getchar();
////////////////////////////////////// Soapy Init ends ////////////////////////////////////////////////////////////

//create a re-usable buffer for rx samples
complex float buf[8192];
uint8_t data[8192];

int j=0; 
float last_phi = 0;
int len=0;

while(1)
{
	fptr=fopen("Samples.bin","rb");//filenameMas4packets
	//fptr=fopen("filename.bin","rb");

	if(fptr==NULL){
	printf("Error \n");
	getchar();
	//break;
	}
	printf("File Opened Sucesfully \n");
  while(1)
  {
	int cnt=fread(&data,1,8192,fptr);

	if(cnt<=0){
		
	printf("File ends \n");
	//getchar();
	break;
	}
	printf("Bytes read=%d \n",cnt);
	int j=0;
	for(int a=0; a< cnt; a=a+2){ // interleaved
	buf[j]=data[a]+data[a+1]*I;
	 j++;
	  //printf("data[a]=%d buf real=%f buf img=%f \n", data[a],crealf(buf[a]),cimagf(buf[a]) );
	}
   
        void *buffs[] = {buf}; //array of buffers
        int flags=0; //flags set by receive operation
        long long timeNs=1000000; //timestamp for receive buffer
      	     
		       int ret= SoapySDRDevice_writeStream(sdr, txStream, &buffs[0], j, &flags, &timeNs, 100000); //100000 
			//ret= SoapySDRDevice_writeStream(txStream, buffs, cnt, &flags, &timeNs, 1000000); //100000 
			//printf("After Read Stream\n");
		       if (ret==SOAPY_SDR_TIMEOUT){
				printf("SOAPY_SDR_TIMEOUT\n");
			continue;
			} 
			else if( ret==SOAPY_SDR_STREAM_ERROR ){
				printf("SOAPY_SDR_STREAM_ERROR\n");
			continue;
			}
			else if( ret==SOAPY_SDR_CORRUPTION){
				printf("SOAPY_SDR_CORRUPTION\n");
			continue;
			}
			else if( ret==SOAPY_SDR_OVERFLOW){
				printf("SOAPY_SDR_OVERFLOW\n");
			continue;
			}
			else if( ret==SOAPY_SDR_NOT_SUPPORTED){
				printf("SOAPY_SDR_NOT_SUPPORTED\n");
			continue;
			}
			else if( ret== SOAPY_SDR_TIME_ERROR){
				printf(" SOAPY_SDR_TIME_ERROR\n");
			continue;
			}
			else if( ret== SOAPY_SDR_UNDERFLOW){
				printf(" SOAPY_SDR_UNDERFLOW\n");
			continue;
			}
			else if (ret>0)
			{
		       	  		
				printf("Sucesfully Sent ret=%d \n",ret);
	
			}

			 if ( SoapySDRDevice_setGain(sdr, SOAPY_SDR_TX, 0, 25.0)==0 )
			   {	
				printf("-----------------------Manual Gain SET \n"); 
			   }
			   else
				{
				printf("Gain Set Error\n");
				exit(0); 
				}

			

 }//while    

fclose(fptr);
printf("content of File sent \n");
getchar(); // send file again 

}
   SoapySDRDevice_deactivateStream(sdr, txStream, 0, 0); //stop streaming
    printf("Done\n");

    return EXIT_SUCCESS;

}

zuckschwerdt, 2024-05-11:
The tx_tools is it's own application with abstractions on backends, you don't need to use any of that.
There is no need to insert any pause, I just tried to emulated your waiting on a keypress.

I'll try you code soon, only thing I noticed is that `complex float buf[8192]` and `uint8_t data[8192]` are not the same size (8192 samples vs 4096 samples), but that should not matter much with the data loop you have.

zuckschwerdt, 2024-05-11:
What you see is not the first data transmission but noise from the setup. Your data transmissions are all "invisible". You need to properly convert your samples to CF32 (or some other format).
Also if your samples are not close to 0 dB you might want to increase the gain.
Also it should be `int ret = SoapySDRDevice_writeStream(sdr, txStream, buffs, j, &flags, timeNs, 100000);`, note the `buffs` and the `timeNs`

zuckschwerdt, 2024-05-11:
Yes, CF32 support in ty_tools is not complete. But CS16 is the native format for the Pluto and for high sample rates (>6M IIRC) CU8 might be better.
Your code does work now, right?

Mas313, 2024-05-12:
YEs. IT works now, Thanks alot. Can you tell about min delay between transmissions if any one has measured before or as per your expereince.

zuckschwerdt, 2024-05-12:
You should be able to send continuously, I don't know of any needed delay.

closed 2024-05-27 by Mas313