pothosware/SoapyPlutoSDR #64: Tx samples Correct format

Mas313, 2024-05-27:
Hi
I am having trouble transmitting with plutosdr with specified formats like CF32 and CS 16 . I have gausian filtered data that has been accumulated (integrated). Now trying to  transmit the data results in no transmisssion, verified through 'SdrSharp'. I want to know what values the soapysdr library expects for pluto to be. When using complex float or 16 bit integer, What is the range of values the data should fit in?. 
Also I have seen example of narrow band fm transmitter where CF32 format is used and the samples values are stored in a variable and assigned to void buffer like this
```
float buff;
--- fill buff samples from wave file
void *buffs={buff}
```
Now transmission is done with writestream function.
Here I assume that wave file already had scaled samples so no scaling was done and wave file transmitted successfully. But it seems strange that CF32 format was specified but samples were stored in float (not complex float) and it did transmit.
Besides this the main question is what are the ranges the data should fit in for float and 16 bit integer samples. 
Thanks

zuckschwerdt, 2024-05-27:
Integer formats always expect the maximum signed range, for 16 bit that would be 2^15-1 = 32767 to -2^15 = -32768.
For float formats the range is normalized to +/-1.0.

The Pluto uses 12 bits but the transfer is 16 bits, so CU8 would leave some dynamic range unused, CS16 could be optimal and CF32 is wasting (USB) bandwith and needs to be converted anyway. See https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Streaming.cpp#L649

Note that complex float is just two channels (I/Q) interleaved floats. If you read from a stereo wave file that maps to the two stereo channels (left/right).

Mas313, 2024-06-04:
I did following but no luck. converted double to int16_t and changed soapy format to CS16

```
 for (int i = 0; i < num_samples; i++) {
        // Scale floating-point sample to signed 16-bit integer range (-32768 to 32767)
        outI[i] = (int16_t)(I[i] * 32767);
        outQ[i] = (int16_t)(Q[i] * 32767);
    }
```

zuckschwerdt, 2024-06-04:
Write your buffers to .cf32 and .cs16 files. Then open with https://triq.org/pdv/ to verify the data. E.g.
```
	int fd = open("foo.cs16", O_WRONLY | O_CREAT | O_TRUNC, 0644);
	if (fd < 0) {
		perror("open()");
		exit(1);
	}
	int r = write(fd, buf, num_samples * 2);
	printf("Wrote: %d bytes\n", r);
	close(fd);
```

Mas313, 2024-06-04:
Thanks for the reply. 
I have attached the results i cannot understood. First screen shot is of file containing I & Q int16_t samples interleaved in a buffer.  
Second screenshot is of complex int16_t buffer with values stores as 
```
buf.real( outI[i]  );
buf.imag( outQ[i] );
```
I am producing zero IF signals so the SDR Fc should have no offset. I dont see any spectrum in any case. 
[image]
[image]

zuckschwerdt, 2024-06-04:
both files look empty. somehow your data is not present in the original or much too quiet (too small numbers).

zuckschwerdt, 2024-06-04:
Try this to fill your complex float buffer. You'll get a pure tone at +10 kHz and a (-27 dB) noise background.
```
    for (int i = 0; i < num_samples; i +=1) {
        float ni = rand() / (float)RAND_MAX * 0.1f - 0.05f;
        float nq = rand() / (float)RAND_MAX * 0.1f - 0.05f;
        float ph = i / 3.0;
        buf[i] = ni + cosf(ph) + nq + sinf(ph) * I;
    }
```

Mas313, 2024-06-05:
I have implemented your code and stores file as
```
int main()
{
	int num_samples=10000;
	
	complex float buf[num_samples];
	
	for (int i = 0; i < num_samples; i +=1) {
        float ni = rand() / (float)RAND_MAX * 0.1f - 0.05f;
        float nq = rand() / (float)RAND_MAX * 0.1f - 0.05f;
        float ph = i / 3.0;
        buf[i] = ni + cosf(ph) + nq + sinf(ph) * I;
    }
    
    int fd = open("Tone.cf32", O_WRONLY | O_CREAT | O_TRUNC, 0644);
	if (fd < 0) {
		perror("open()");
		exit(1);
	}
	int r = write(fd, buf, num_samples * sizeof(float) * 2);
	printf("Wrote: %d bytes\n", r);
	close(fd);
    
    return 0;
}

Got following
![TonefloatComplex](https://github.com/pothosware/SoapyPlutoSDR/assets/92240144/22d26c79-dfeb-4f6f-941c-060cec6fe298)

```

zuckschwerdt, 2024-06-05:
Looks like a good transmission! The harmonics would be from the maxed out amplitude, you can try to scale the cos and sin down with e.g. `0.9 *`
This example shows that very likely your earlier input buffer did not contain good samples.

closed 2024-06-19 by Mas313