pothosware/SoapyPlutoSDR #67: Cross compiling SoapyPlutoSDR

Mas313, 2024-08-22:
When compiling for pluto arm processor I get error that soapysdr configuration file is not compatible. Although SoapySDR for arm has been built successfully. 
For building SoapyPlutoSDR I used following commands
```
cmake \
    -D CMAKE_C_COMPILER=/usr/bin/arm-linux-gnueabihf-gcc \
    -D CMAKE_CXX_COMPILER=/usr/bin/arm-linux-gnueabihf-g++ \
    -D CMAKE_AR=/usr/bin/arm-linux-gnueabihf-ar \
    -D CMAKE_C_COMPILER_AR=/usr/bin/arm-linux-gnueabihf-gcc-ar \
    -D CMAKE_C_COMPILER_RANLIB=/usr/bin/arm-linux-gnueabihf-gcc-ranlib \
    -D CMAKE_LINKER=/usr/bin/arm-linux-gnueabihf-ld \
    -D CMAKE_NM=/usr/bin/arm-linux-gnueabihf-nm \
    -D CMAKE_OBJCOPY=/usr/bin/arm-linux-gnueabihf-objcopy \
    -D CMAKE_OBJDUMP=/usr/bin/arm-linux-gnueabihf-objdump \
    -D CMAKE_RANLIB=/usr/bin/arm-linux-gnueabihf-ranlib \
    -D CMAKE_STRIP=/usr/bin/arm-linux-gnueabihf-strip \
    ..
make
sudo make install
```

zuckschwerdt, 2024-08-22:
Can you elaborate? What file is not compatible? Do you encounter this while compiling or while running? What commands did you try?

I guess it would be a problem that you didn't use a sysroot. And I'd use cmake's toolchain option.

There are PlutoSDR binaries for SoapySDR, rtl_433 and some tools available here: https://github.com/triq-org/PlutoSDR-Apps/ -- and you can see an example cross compile setup.

closed 2024-08-22 by zuckschwerdt

Mas313, 2024-08-22:
I got this error

CMake Warning at CMakeLists.txt:14 (find_package):
  Could not find a configuration file for package "SoapySDR" that is
  compatible with requested version "".

  The following configuration files were considered but not accepted:

    /usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake, version: 0.8.1-unknown (64bit)

CMake Error at CMakeLists.txt:16 (message):
  Soapy SDR development files not found...

-- Configuring incomplete, errors occurred!

zuckschwerdt, 2024-08-22:
Seems like you are missing a sysroot with cross-compiled SoapySDR.

Mas313, 2024-08-22:
I have added path to SoapySDR lib that has been succesfully built for arm, but still i get error
-D CMAKE_LIBRARY_PATH=/home/csg/masood/pluto-0.30.sysroot/usr/lib/SoapySDR \

zuckschwerdt, 2024-08-22:
Adding a lib path might not suffice, I still recommend a full sysroot with all the support files like pkgconfig and such. Look into cmake toolchains, e.g. https://cmake.org/cmake/help/latest/manual/cmake-toolchains.7.html#cross-compiling-for-linux