pothosware/SoapyPlutoSDR #69: More accurate documents to build the SoapyPlutoSDR on Windows VS 2022 MSVC17

DCZYewen, 2024-10-17:
` cmake .. -G "Visual Studio 17 2022" -A x64 -DLibIIO_INCLUDE_DIR="C:\\Program Files (x86)\\Microsoft Visual Studio 12.0\\VC\\include" -DLibIIO_LIBRARY="C:\\Program Files (x86)\\Microsoft Visual Studio 12.0\\VC\\lib\\amd64"  -DLibAD9361_INCLUDE_DIR="C:\\Program Files (x86)\\Microsoft Visual Studio 12.0\\VC\\include" -DLibAD9361_LIBRARY="C:\\Program Files (x86)\\Microsoft Visual Studio 12.0\\VC\\lib\\amd64" `
 
 I cloned this library on my pc, the explicitly points out where is libiio and libad9361, but CMake said as below. As far as i know, that Windows uses WinUSB instead of libusb-1.0 as its USB devices intereacting library. and there is no pkg-config support on Windows.

 ```
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Found LibAD9361: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64
-- LibAD9361_INCLUDE_DIRS: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/include
-- LibAD9361_LIBRARIES: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64
-- LibAD9361_DEFINITIONS:
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find LibUSB (missing: LibUSB_LIBRARY LibUSB_INCLUDE_DIR)
-- Module PlutoSDRSupport configured with version: 0.2.2
-- Configuring done (0.2s)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY (ADVANCED)
    linked by target "PlutoSDRSupport" in directory C:/Users/lzhal/Desktop/SoapyPlutoSDR-master

CMake Warning at C:/Program Files/SoapySDR/cmake/SoapySDRUtil.cmake:82 (target_link_libraries):
  Target "PlutoSDRSupport" requests linking to directory "C:/Program Files
  (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64".  Targets may link only to
  libraries.  CMake is dropping the item.
Call Stack (most recent call first):
  CMakeLists.txt:81 (SOAPY_SDR_MODULE_UTIL)

CMake Warning at C:/Program Files/SoapySDR/cmake/SoapySDRUtil.cmake:82 (target_link_libraries):
  Target "PlutoSDRSupport" requests linking to directory "C:/Program Files
  (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64".  Targets may link only to
  libraries.  CMake is dropping the item.
Call Stack (most recent call first):
  CMakeLists.txt:81 (SOAPY_SDR_MODULE_UTIL)

CMake Warning at C:/Program Files/SoapySDR/cmake/SoapySDRUtil.cmake:82 (target_link_libraries):
  Target "PlutoSDRSupport" requests linking to directory "C:/Program Files
  (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64".  Targets may link only to
  libraries.  CMake is dropping the item.
Call Stack (most recent call first):
  CMakeLists.txt:81 (SOAPY_SDR_MODULE_UTIL)

CMake Warning at C:/Program Files/SoapySDR/cmake/SoapySDRUtil.cmake:82 (target_link_libraries):
  Target "PlutoSDRSupport" requests linking to directory "C:/Program Files
  (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64".  Targets may link only to
  libraries.  CMake is dropping the item.
Call Stack (most recent call first):
  CMakeLists.txt:81 (SOAPY_SDR_MODULE_UTIL)

CMake Warning at C:/Program Files/SoapySDR/cmake/SoapySDRUtil.cmake:82 (target_link_libraries):
  Target "PlutoSDRSupport" requests linking to directory "C:/Program Files
  (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64".  Targets may link only to
  libraries.  CMake is dropping the item.
Call Stack (most recent call first):
  CMakeLists.txt:81 (SOAPY_SDR_MODULE_UTIL)

CMake Warning at C:/Program Files/SoapySDR/cmake/SoapySDRUtil.cmake:82 (target_link_libraries):
  Target "PlutoSDRSupport" requests linking to directory "C:/Program Files
  (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64".  Targets may link only to
  libraries.  CMake is dropping the item.
Call Stack (most recent call first):
  CMakeLists.txt:81 (SOAPY_SDR_MODULE_UTIL)

CMake Warning at C:/Program Files/SoapySDR/cmake/SoapySDRUtil.cmake:82 (target_link_libraries):
  Target "PlutoSDRSupport" requests linking to directory "C:/Program Files
  (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64".  Targets may link only to
  libraries.  CMake is dropping the item.
Call Stack (most recent call first):
  CMakeLists.txt:81 (SOAPY_SDR_MODULE_UTIL)

CMake Warning at C:/Program Files/SoapySDR/cmake/SoapySDRUtil.cmake:82 (target_link_libraries):
  Target "PlutoSDRSupport" requests linking to directory "C:/Program Files
  (x86)/Microsoft Visual Studio 12.0/VC/lib/amd64".  Targets may link only to
  libraries.  CMake is dropping the item.
Call Stack (most recent call first):
  CMakeLists.txt:81 (SOAPY_SDR_MODULE_UTIL)

-- Generating done (0.1s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

DCZYewen, 2024-10-17:
problem solved, should i make a pull request?

zuckschwerdt, 2024-10-17:
I was going to say this doesn't sound right. LibUSB ist optional (https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Registration.cpp#L6) and only linked when available (https://github.com/pothosware/SoapyPlutoSDR/blob/master/CMakeLists.txt#L70). How did you solve the problem?

DCZYewen, 2024-10-17:
How can i edit the wiki, i cloned the wiki repository and i dont have the privilege to edit it.

DCZYewen, 2024-10-17:
yep  it seems that SoapyPlutoSDR is not necessarly rely on libusb, but the cmake files says it is.

DCZYewen, 2024-10-17:
still cant figure out how to modify the wiki, but i managed to get it work.
[image]

zuckschwerdt, 2024-10-17:
Do you mean the main https://github.com/pothosware/SoapyPlutoSDR/wiki? I don't think it's locked.

Do you mean because of https://github.com/pothosware/SoapyPlutoSDR/blob/master/CMakeLists.txt#L87 -- I think I'll wrap the use of `${LibUSB_LIBRARIES}` and `${LibAD9361_LIBRARIES}` there.

DCZYewen, 2024-10-17:
i cloned the https://github.com/pothosware/SoapyPlutoSDR.wiki.git and failed to edit the repo.

zuckschwerdt, 2024-10-17:
Should work now. Thanks!

Closed by: 94889f7

closed 2024-10-17 by zuckschwerdt