pothosware/SoapyPlutoSDR #71: LibreSDR context selection with Soapy possible? - (Forcing to use the Ethernet interface instead of USB)

computer-wizard, 2025-04-13:
The LibreSDR is a souped up Pluto compatible with a larger FPGA (Z7020) and an Ethernet internet interface.

iio_info -s  gives,

Available contexts:
	0: 192.168.0.12 (LibreSDR Rev.5 (Z7020-AD9361)), serial=3KVVBQVFYT5H6VRN [ip:libresdr.local]
	1: 192.168.2.1 (LibreSDR Rev.5 (Z7020-AD9361)), serial=3KVVBQVFYT5H6VRN [ip:libresdr.local]
	2: (iwlwifi_1,jc42,acpitz,coretemp,jc42 on Intel Corporation) [local:]
	3: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=3KVVBQVFYT5H6VRN [usb:2.5.5]

Contexts 0, 1 and 3 are of the LibreSDR

If I use the Soapy driver it uses context 3, the USB device for the data.

Question:

Can I force Soapy to use context 0, what is the Ethernet interface IP as that has a  much higher throughput?

zuckschwerdt, 2025-04-13:
Try `uri=ip:192.168.0.12` as option.

computer-wizard, 2025-04-13:
SoapySDRUtil --make=driver=plutosdr,uri=ip:192.168.0.12
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device driver=plutosdr,uri=ip:192.168.0.12
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9361
  ad9361-phy,xo_correction=40000000
  backend_version=0.25 (git tag: v0.25)
  fw_version=tezuka-0.1.3
  hw_model=LibreSDR Rev.5 (Z7020-AD9361)
  hw_model_variant=1
  hw_serial=3KVVBQVFYT5H6VRN
  ip,ip-addr=192.168.0.12
  library_version=0.25 (git tag: v0.25)
  local,kernel=6.1.0
  uri=ip:192.168.0.12

It is sadly still routed over USB when I monitor it :-(

zuckschwerdt, 2025-04-14:
You can also try `hostname=192.168.0.12` that should directly create a network context. Otherwise all possible contexts are scanned and filter by uri, see find_PlutoSDR at https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Registration.cpp#L11
The actuall context creation (uri, hostname, or default) is in the SoapyPlutoSDR constructor at https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Settings.cpp#L19-L27